Josverl/micropython-stubber #4: module.json schema is too simplistic

Josverl, 2019-06-30:
the module .json file uses a very simplistic format ( a single array of things) , making it hard to read and reason over it.

a better format would be something such as : 
```json
{
    "firmware": {
        "machine": "ESP32 board with ESP32",
        "firmware": "1.2.3.4",
        "nodename": "esp32_LoBo",
        "version": "3.2.24",
        "release": "3.2.24",
        "sysname": "esp32_LoBo"
    },
    "stubber": {
        "version": "1.2.1"
    },
    "modules": [
        {
            "file": "file_name",
            "module": "module_name"
        },
        {
            "file": "file_name",
            "module": "module_name"
        },
        {
            "file": "file_name",
            "module": "module_name"
        }
    ]
}
```

Josverl, 2019-06-30:
@BradenM ,

I have implemented the above change in the json format starting with version 1.2.0 
i think that should improve the parsing as formally a json array is un-ordered so skipping the first 2 nodes was a bit of a hack.

Could  you take a look at commit cb0779b  and let me know if that works for you ?

/jos

BradenM, 2019-07-02:
Looks good to me. Definitely agree with getting rid of the array, and the firmware specs should be much easier to parse now.

I am not and won't be home where I can test this out on some more devices/firmware for the rest of this week, but I will definitely give it a more detailed look when I get back.

Josverl, 2019-07-02:
NP,
I’ll push it to main, and we can continue to adopt as needed.

enjoy the fireworks

/jos

BradenM, 2019-07-08:
@Josverl,

Was finally able to check this out properly. Definitely a lot more useful, but I am still struggling with the issue of being able to determine exactly what branch of micropython a device is using.

I am trying to avoid hard coding it, as according to [awesome-micropython](https://github.com/pfalcon/awesome-micropython), there are over 2000 of 'em. So far I haven't been able to come up with anything better than cross-referencing the version number against the tags/ports from a list of micropython git repos (See [this](https://github.com/BradenM/micropy-stubs/blob/2d1b16087d126cd5b5de90829c1c945a7bcd6bd5/main.py#L202) and [its output](https://github.com/BradenM/micropy-stubs/blob/2d1b16087d126cd5b5de90829c1c945a7bcd6bd5/packages/micropython-official/info.json#L37)). Not exactly a perfect approach.

For `micropy-cli`, the benefit of knowing the exact firmware would make it pretty easy to determine which (firmware) frozen modules to retrieve and source in a project. I may just be over complicating this haha, but do you happen to have any ideas? Thanks.

BradenM, 2019-07-09:
@Josverl,

For now this schema works great, I have gone ahead and integrated it into `micropy-cli`. I have pushed tackling firmware resolution to a later date, so when you have the time do you mind removing the `stubber.modules` debug line and merging into master? 

Thanks for your help!

Josverl, 2019-07-10:
I've merged the schema to main.

i agree that all the different firmware options make it hard to find the capabilities of the language , and of the included optional modules; especially as these modules also can/are sometimes changed for the specific firmware.

closed 2019-07-10 by Josverl