Josverl/micropython-stubber #5: get the number of positionnal arguments or variable

pmp-p, 2019-07-10:
following https://github.com/microsoft/vscode-python/issues/5098 i had an idea (won't work on closed source, but who would like to work with a closed source micropython anyway ) ?

Maybe try that way:
  rgrep  ^STATIC MP_DEFINE_CONST_FUN_OBJ on all the source tree

per line you'll have
1)  will be name of module ***** generally py/mod******.c

2)  you will get either  MP_DEFINE_CONST_FUN_OBJ_x where x is the number of fixed args
or MP_DEFINE_CONST_FUN_OBJ_VAR for variable arguments list.

3) you'll get a () couple where the 2nd item is the function name
so i guess you are extracting from a live board  and according to the example i gave you in the wrong place (sorry for that) you've closed to my nose when i was about to apologize.

you would  get :
```
py/modmicropython.c:STATIC MP_DEFINE_CONST_FUN_OBJ_2(mp_micropython_schedule_obj, mp_micropython_schedule);```

that means micropython.schedule(arg0, arg1) which is probably better than micropython.schedule()

also that also gives name of relevant C source, maybe more can be extracted there
eg ```rgrep mp_micropython_schedule|grep { |grep -v MP_ROM```
could inform that :
```micropython.schedule(function, arg)```

pmp-p, 2019-07-10:
and no i still don't have finished reading documentation yet, but i will try to

Josverl, 2019-07-10:
thanks for the suggestion, 
this was my earliest thinking as well, and as far as I can tell several others have tried that ( see \tools\gendoc.py  which is part of micropython) but that also will require to (re)build or integrate a multiplatform C pre-processor to deal with all the C prepocessor conditioonals in order to select the correct lines from the sourcefiles. 
and while that would bee very cool , that is also is a lot of work. 

perhaps there is a way to 
1) run the precompiler  with the correct options ( likely needs a make file as well ) 
2) filter for the ^STATIC MP_DEFINE_CONST_FUN_OBJ '
3) deal with the duplicates
4) attribute that to a specific firmware build

pmp-p, 2019-07-10:
i think the qstr process do all the preprocessing stuff you need but then you do not have access to sources but to *a build*

Josverl, 2019-07-10:
given the number of processors and OSs , there is no such thing as native to all hosts ...
and i would not want to take a dependency on all toolchains for all ports ... on all platforms either ,
lets get started with as simple a solution as can be.
can you create a concept script that extracts the relevant information assuming a git clone has been done ?
lets pick the network module or any other that you prefer and lets see how far we can get.

pmp-p, 2019-07-10:
>  i would not want to take a dependency on all toolchains for all ports ... on all platforms either

you only need clang -E that should not be hard to get, 

you can have both  node and clang packed with emscripten portable SDK  which was considered here https://github.com/microsoft/vscode-python/issues/2375  ( could even run mock module in there with some visual display if using micropython wasm )

> can you create a concept script that extracts the relevant information assuming a git clone has been done 

not i'm actually more interested in the other way around : 
parsing python annotations to create C modules headers for people with python abilities who can't write C.

Josverl, 2019-07-10:
so what is your suggested approach, emscripten ?
I see no benefit to microstubber for any/all micropython sourcecode code to be be transpiled using emscripten ?

What is the expected success ratio for that ? I see quite a few red flags mentionedin https://emscripten.org/docs/porting/guidelines/portability_guidelines.html#code-portability-guidelines

you can have both node and clang packed with emscripten portable SDK which was considered here microsoft/vscode-python#2375<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicrosoft%2Fvscode-python%2Fissues%2F2375&data=02%7C01%7C%7C77e1951d381f4c51c17a08d70577f4bd%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636983885065834136&sdata=WGleaSOW6e2%2B8bgl9ZXMnXcMhPZR1qtUj7fM%2Bw6jlsM%3D&reserved=0> ( could even run mock module in there with some visual display if using micropython wasm )

pmp-p, 2019-07-10:
no i meant using the emscripten sdk tools for getting hand on a valid preprocessor for all platforms,
you can preprocess ** all ports ** from the same git source with the included tools and they are easy to install for any platform.

Josverl, 2020-10-18:
ivé tries the c-route , it only got me a very very limited number of module primitives/stubs  that were useful by using regex pattern matching.
specifically the matches depend very much on the coding styles and naming conventions used. due to the OSS nature with many authors , and very different tools, i do not think this is worth the effort.
partial work in this branch https://github.com/Josverl/micropython-stubber/commits/feat/stub_from_source
[getfromsource.ps1 script to parse source](https://github.com/Josverl/micropython-stubber/blob/feat/stub_from_source/runOnPc/getfromsource.ps1)

these are

closed 2020-10-18 by Josverl