Josverl/micropython-stubber #7: Add support for stubbing nested modules

Josverl, 2019-08-30:
add support for stubbing nested modules on LEGO and M5STACK 

see https://github.com/BradenM/micropy-cli/issues/43

Josverl, 2019-09-01:
@ vbolshakov 

I have made a number of improvements to deal with importing nested packages of frozen modules.
I have tested this using the m5stack firmware , and that works quite well with [createstubs](https://github.com/Josverl/micropython-stubber/blob/master/createstubs.py) version 1.3.1 or newer 

For the LEGO EV3 brick you should able to use that same version but you will need to make a few changes. 

Add a line / lines to add your specific modules: 
``` python
# almost at the end of the file
def main():
    stubber = Stubber(firmware_id='LEGO EV3 v1.0.0')
    stubber.add_modules([ 'builtins', 'cmath', 'ev3brick_c', 'ev3devices_c', 
        'mmap', 'parameters_c', 'pybricks/display', 'pybricks/ev3brick', 'pybricks/ev3devices', 'pybricks/ev3devio', 'pybricks/init', 'pybricks/parameters', 
        'pybricks/robotics', 'pybricks/speaker', 'pybricks/tools', 'pybricks/uev3dev/_alsa', 'pybricks/uev3dev/_wand', 'pybricks/uev3dev/display', 'pybricks/uev3dev/i2c', 
        'pybricks/uev3dev/init', 'pybricks/uev3dev/messaging', 'pybricks/uev3dev/sound', 'pybricks/uev3dev/util', 'sys', 'termios', 'tools', 'umachine'])

```
also see : https://github.com/Josverl/micropython-stubber#custom-firmware

**one other thing** : it seems that the LEGO brick does nothave the logging module.
in that case I have added a copy of that [logging module](https://github.com/Josverl/micropython-stubber/blob/master/logging.py). 
you should add that to the EV3 before `import createstubs`

vbolshakov, 2019-09-03:
@Josverl New version not work at all.. Previous made stubs for standard modules like "os". New unable to. If you interesting, i can provide ssh to EV3.

BradenM, 2019-09-04:
@vbolshakov,

You mind providing the log output? Although not with EV3, I have tested the latest `createstubs.py` on a few different firmware and it doesn't seem to have an issue.

Thanks.

vbolshakov, 2019-09-04:
I modify copy createstubs.py and logging.py to EV3. Modify createstubs.py - in several places - there is differences in os.uname. Than replace end of file to this:
```
def main():
    try:
        logging.basicConfig(level=logging.INFO)
    except:
        pass
    stubber = Stubber(firmware_id='LEGO EV3 v1.0.0')
    stubber.add_modules(['pybricks/display', 'pybricks/ev3brick', 'pybricks/ev3devices'])
    stubber.create_all_stubs()
    stubber.report()

main()
```
On output i got this:
```
robot@ev3dev:~$ pybricks-micropython ./createstubs.py
INFO:stubber:Start micropython-stubber v1.3.2 on LEGO EV3 v1.0.0
INFO:stubber:Stub module: pybricks/display     to file: /home/robot/stubs/LEGO_EV3_v1_0_0/pybricks/display.py   mem:993536
INFO:stubber:Stub module: pybricks/ev3brick    to file: /home/robot/stubs/LEGO_EV3_v1_0_0/pybricks/ev3brick.py  mem:971360
INFO:stubber:Stub module: pybricks/ev3devices  to file: /home/robot/stubs/LEGO_EV3_v1_0_0/pybricks/ev3devices.py mem:869488
INFO:stubber:Stub module: uasyncio/core        to file: /home/robot/stubs/LEGO_EV3_v1_0_0/uasyncio/core.py      mem:867248
INFO:stubber:Stub module: umqtt/robust         to file: /home/robot/stubs/LEGO_EV3_v1_0_0/umqtt/robust.py       mem:854960
INFO:stubber:Stub module: umqtt/simple         to file: /home/robot/stubs/LEGO_EV3_v1_0_0/umqtt/simple.py       mem:849936
INFO:stubber:Stub module: urllib/urequest      to file: /home/robot/stubs/LEGO_EV3_v1_0_0/urllib/urequest.py    mem:849936
INFO:stubber:Stub module: _thread              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/_thread.py            mem:848352
INFO:stubber:Stub module: ak8963               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/ak8963.py             mem:848352
INFO:stubber:Stub module: apa102               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/apa102.py             mem:848352
INFO:stubber:Stub module: apa106               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/apa106.py             mem:848352
INFO:stubber:Stub module: array                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/array.py              mem:848352
INFO:stubber:Stub module: binascii             to file: /home/robot/stubs/LEGO_EV3_v1_0_0/binascii.py           mem:848352
INFO:stubber:Stub module: btree                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/btree.py              mem:843984
INFO:stubber:Stub module: builtins             to file: /home/robot/stubs/LEGO_EV3_v1_0_0/builtins.py           mem:843984
INFO:stubber:Stub module: cmath                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/cmath.py              mem:843984
INFO:stubber:Stub module: collections          to file: /home/robot/stubs/LEGO_EV3_v1_0_0/collections.py        mem:843968
INFO:stubber:Stub module: crypto               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/crypto.py             mem:841632
INFO:stubber:Stub module: curl                 to file: /home/robot/stubs/LEGO_EV3_v1_0_0/curl.py               mem:841632
INFO:stubber:Stub module: dht                  to file: /home/robot/stubs/LEGO_EV3_v1_0_0/dht.py                mem:841632
INFO:stubber:Stub module: display              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/display.py            mem:841632
INFO:stubber:Stub module: ds18x20              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/ds18x20.py            mem:841632
INFO:stubber:Stub module: errno                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/errno.py              mem:841632
INFO:stubber:Stub module: esp                  to file: /home/robot/stubs/LEGO_EV3_v1_0_0/esp.py                mem:841632
INFO:stubber:Stub module: esp32                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/esp32.py              mem:841632
INFO:stubber:Stub module: flashbdev            to file: /home/robot/stubs/LEGO_EV3_v1_0_0/flashbdev.py          mem:841632
INFO:stubber:Stub module: framebuf             to file: /home/robot/stubs/LEGO_EV3_v1_0_0/framebuf.py           mem:841632
INFO:stubber:Stub module: freesans20           to file: /home/robot/stubs/LEGO_EV3_v1_0_0/freesans20.py         mem:841632
INFO:stubber:Stub module: functools            to file: /home/robot/stubs/LEGO_EV3_v1_0_0/functools.py          mem:841632
INFO:stubber:Stub module: gc                   to file: /home/robot/stubs/LEGO_EV3_v1_0_0/gc.py                 mem:840928
INFO:stubber:Stub module: gsm                  to file: /home/robot/stubs/LEGO_EV3_v1_0_0/gsm.py                mem:840928
INFO:stubber:Stub module: hashlib              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/hashlib.py            mem:840928
INFO:stubber:Stub module: heapq                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/heapq.py              mem:816752
INFO:stubber:Stub module: inisetup             to file: /home/robot/stubs/LEGO_EV3_v1_0_0/inisetup.py           mem:813280
INFO:stubber:Stub module: io                   to file: /home/robot/stubs/LEGO_EV3_v1_0_0/io.py                 mem:813280
INFO:stubber:Stub module: json                 to file: /home/robot/stubs/LEGO_EV3_v1_0_0/json.py               mem:813168
INFO:stubber:Stub module: logging              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/logging.py            mem:794976
INFO:stubber:Stub module: lwip                 to file: /home/robot/stubs/LEGO_EV3_v1_0_0/lwip.py               mem:794976
INFO:stubber:Stub module: machine              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/machine.py            mem:794976
[... cut ...]
INFO:stubber:Stub module: writer               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/writer.py             mem:779600
INFO:stubber:Stub module: ymodem               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/ymodem.py             mem:779600
INFO:stubber:Stub module: zlib                 to file: /home/robot/stubs/LEGO_EV3_v1_0_0/zlib.py               mem:779600
INFO:stubber:Finally done
INFO:stubber:Created stubs for 0 modules on board LEGO EV3 v1.0.0
Path: /home/robot/stubs/LEGO_EV3_v1_0_0
```
And all generated files are empty.

BradenM, 2019-09-04:
Strange. Sounds like a memory related issue to me, but it seems like you still had a good amount free...

You'll definitely have to wait until @Josverl can take a look, cause I'm not sure what the issue could be.

Josverl, 2019-09-04:
thanks for sharing the log , that seems to indicate the module goes though the motions, 
but perhaps something is preventing it from creating the files ?

i see that the path is deeper than i normally see on mpy devices ( is rather than /stubs or /flash/stubs there is an additional level with /home/robot/stubs

lets start by cranking up to logging by changing :   
`ogging.basicConfig(level=logging.INFO)`   
to :   
ogging.basicConfig(level=logging.DEBUG)

that will give us more detail on what is happening during the processing.

**specify patch**

if the path is the problem ( speculation at this point) then it is possible to specify the path 
by specifying it like :
`    stubber = Stubber(path='/home/stubs', firmware_id='LEGO EV3 v1.0.0')` 

i do not know what paths are writable, or  if this allows you to write to a flash card  mounted on '/sd' using something like :  
`    stubber = Stubber(PATH='/sd/stubs',firmware_id='LEGO EV3 v1.0.0')`

vbolshakov, 2019-09-04:
Previous version create files for "normal" modules without any problems. This version create empty files.
This is debug:
```
robot@ev3dev:~$ pybricks-micropython ./createstubs.py
INFO:stubber:Start micropython-stubber v1.3.2 on LEGO EV3 v1.0.0
INFO:stubber:Stub module: pybricks/display     to file: /home/robot/stubs/LEGO_EV3_v1_0_0/pybricks/display.py   mem:993552
DEBUG:stubber:Memory     :               993552  22192
INFO:stubber:Stub module: pybricks/ev3brick    to file: /home/robot/stubs/LEGO_EV3_v1_0_0/pybricks/ev3brick.py  mem:971360
DEBUG:stubber:Memory     :               971360 101872
INFO:stubber:Stub module: pybricks/ev3devices  to file: /home/robot/stubs/LEGO_EV3_v1_0_0/pybricks/ev3devices.py mem:869488
DEBUG:stubber:Memory     :               869488   2240
INFO:stubber:Stub module: uasyncio/core        to file: /home/robot/stubs/LEGO_EV3_v1_0_0/uasyncio/core.py      mem:867248
DEBUG:stubber:Memory     :               867248  12288
INFO:stubber:Stub module: umqtt/robust         to file: /home/robot/stubs/LEGO_EV3_v1_0_0/umqtt/robust.py       mem:854960
DEBUG:stubber:Memory     :               854960   5024
INFO:stubber:Stub module: umqtt/simple         to file: /home/robot/stubs/LEGO_EV3_v1_0_0/umqtt/simple.py       mem:849936
DEBUG:stubber:Memory     :               849936      0
INFO:stubber:Stub module: urllib/urequest      to file: /home/robot/stubs/LEGO_EV3_v1_0_0/urllib/urequest.py    mem:849936
DEBUG:stubber:Memory     :               849936   1600
INFO:stubber:Stub module: _thread              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/_thread.py            mem:848352
DEBUG:stubber:Memory     :               848352      0
INFO:stubber:Stub module: ak8963               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/ak8963.py             mem:848352
DEBUG:stubber:Failed to import module: ak8963
DEBUG:stubber:Memory     :               848352      0
INFO:stubber:Stub module: apa102               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/apa102.py             mem:848352
DEBUG:stubber:Failed to import module: apa102
DEBUG:stubber:Memory     :               848352      0
INFO:stubber:Stub module: apa106               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/apa106.py             mem:848352
DEBUG:stubber:Failed to import module: apa106
DEBUG:stubber:Memory     :               848352      0
INFO:stubber:Stub module: array                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/array.py              mem:848352
DEBUG:stubber:Memory     :               848352      0
INFO:stubber:Stub module: binascii             to file: /home/robot/stubs/LEGO_EV3_v1_0_0/binascii.py           mem:848352
DEBUG:stubber:Memory     :               848352   4368
INFO:stubber:Stub module: btree                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/btree.py              mem:843984
DEBUG:stubber:Memory     :               843984      0
INFO:stubber:Stub module: builtins             to file: /home/robot/stubs/LEGO_EV3_v1_0_0/builtins.py           mem:843984
DEBUG:stubber:Memory     :               843984      0
INFO:stubber:Stub module: cmath                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/cmath.py              mem:843984
DEBUG:stubber:Memory     :               843984      0
INFO:stubber:Stub module: collections          to file: /home/robot/stubs/LEGO_EV3_v1_0_0/collections.py        mem:843968
DEBUG:stubber:Memory     :               843968   2352
INFO:stubber:Stub module: crypto               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/crypto.py             mem:841632
DEBUG:stubber:Failed to import module: crypto
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: curl                 to file: /home/robot/stubs/LEGO_EV3_v1_0_0/curl.py               mem:841632
DEBUG:stubber:Failed to import module: curl
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: dht                  to file: /home/robot/stubs/LEGO_EV3_v1_0_0/dht.py                mem:841632
DEBUG:stubber:Failed to import module: dht
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: display              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/display.py            mem:841632
DEBUG:stubber:Failed to import module: display
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: ds18x20              to file: /home/robot/stubs/LEGO_EV3_v1_0_0/ds18x20.py            mem:841632
DEBUG:stubber:Failed to import module: ds18x20
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: errno                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/errno.py              mem:841632
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: esp                  to file: /home/robot/stubs/LEGO_EV3_v1_0_0/esp.py                mem:841632
DEBUG:stubber:Failed to import module: esp
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: esp32                to file: /home/robot/stubs/LEGO_EV3_v1_0_0/esp32.py              mem:841632
DEBUG:stubber:Failed to import module: esp32
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: flashbdev            to file: /home/robot/stubs/LEGO_EV3_v1_0_0/flashbdev.py          mem:841632
DEBUG:stubber:Failed to import module: flashbdev
DEBUG:stubber:Memory     :               841632      0
INFO:stubber:Stub module: framebuf             to file: /home/robot/stubs/LEGO_EV3_v1_0_0/framebuf.py           mem:841632
[... cut ...]
INFO:stubber:Stub module: ymodem               to file: /home/robot/stubs/LEGO_EV3_v1_0_0/ymodem.py             mem:779600
DEBUG:stubber:Failed to import module: ymodem
DEBUG:stubber:Memory     :               779600      0
INFO:stubber:Stub module: zlib                 to file: /home/robot/stubs/LEGO_EV3_v1_0_0/zlib.py               mem:779600
DEBUG:stubber:Memory     :               779600     64
INFO:stubber:Finally done
INFO:stubber:Created stubs for 0 modules on board LEGO EV3 v1.0.0
Path: /home/robot/stubs/LEGO_EV3_v1_0_0
```

vbolshakov, 2019-09-04:
Can give you ssh access by sending credentials to your email.

BradenM, 2019-09-04:
@vbolshakov, I could take a stab at it if Josverl isn't available right now. 

You can reach me at bradenmars@bradenmars.me if that sounds fine.

BradenM, 2019-09-04:
@vbolshakov,

Thanks, got it working. See #9

vbolshakov, 2019-09-04:
Thx! Great work!

Josverl, 2019-09-05:
Thanks for identifying and fixing this while I was flying across europe.
try/except is a blessing and a pain for unexpected errors.

Ill integrate the fix for the absent uname(); and I like the idea of being able to use the linux port to do some testing.  
I found building stable and repeatable and unit test against hardware quite challenging.

Josverl, 2020-10-18:
moved preview to master

closed 2020-10-18 by Josverl