Josverl/micropython-stubber #13: Add Bluetooth Library

askpatrickw, 2020-05-15:
Is there any concern with adding bluetooth\ubluetooth to createstubs?

I'm happy to submit a PR, but I wanted to check that all is needed is to add it to the hardcoded list starting around line 80.

This would be my PR
https://github.com/askpatrickw/micropython-stubber/commit/7fba58e7362f62a4759f5aa11466e4d0373de492#diff-f980488c64e09b41b963829d88a15d94

Josverl, 2020-07-25:
Hi Patrick , 
sorry for the tardiness 
yes that is one of the ways that should allow the stubs to be generated for the bluetooth module.
another way is to uncomment & change [one line](https://github.com/Josverl/micropython-stubber/blob/ccd73538dde8a5da77b89aba79c0d7a076e5eecd/createstubs.py#L384) in the script 

``` Python
    # stubber.add_modules(['xyz'])  TO 
   stubber.add_modules(['bluetooth'])

```

for which firmware are you generating ?
if it is a common firmware i'm happy to merge your PR

askpatrickw, 2020-09-11:
Hi @Josverl sorry long in getting back to you, this was on MicroPython on a TinyPico (If I remember correctly) not a fork.

Josverl, 2020-10-18:
Hi Patrick , 
I have updated the code and the [version 1.3.4](https://github.com/Josverl/micropython-stubber/blob/master/board/createstubs.py) includes the stubbing of the Bluetooth module if it is included in the firmware.

i've ran the stubber on a nightly of the 1.13 build and you can pick up the [bluetooth](https://github.com/Josverl/micropython-stubs/blob/master/esp32_1_13_0-103/bluetooth.py) module on the [micropython-stubs](https://github.com/Josverl/micropython-stubs) repo

Thanks for pointing this out.

closed 2020-10-18 by Josverl