Josverl/micropython-stubber #42: create stubs for ulab

Josverl, 2020-10-31:
https://github.com/v923z/micropython-ulab

 [ ] create build that includes ulab version x
 [ ] run createstubs 
 [ ] add to stubs repo

CallumJHays, 2020-12-05:
G'day I've given this a go with both `micropy-cli` and running `create_stubs.py` through rshell, but no stubs for ulab are created.

[`ulab` itself appears to be able to generate stubs with rich type info with its docs](https://github.com/v923z/micropython-ulab/blob/13ce751255a012b518605a95cd3ef83cafeda661/docs/manual/Makefile#L18); I'm wondering if we could include this functionality somehow in either this repo, [micropython-stubs](https://github.com/Josverl/micropython-stubs) or [micropy-cli](https://github.com/BradenM/micropy-cli).

In the meantime I'm going to add it into my project manually.

Josverl, 2020-12-06:
Did you add ulab as a module to generate?
You will need to do this manually as there is no way in micropython to get a list of available modules in a given firmware.

Or if ulab is a module included in one of the common builds than Im happy to add it to the core list of modules to try

CallumJHays, 2020-12-08:
> Did you add ulab as a module to generate?
You will need to do this manually as there is no way in micropython to get a list of available modules in a given firmware.

Ah no I haven't - didn't know that that was the case. I'll give that a try later today and get back to you.

> Or if ulab is a module included in one of the common builds than Im happy to add it to the core list of modules to try.

Afaik the only builds with it inbuilt are openmv and circuitpython - could these be supported? It would actually be very useful for users of my project so I'll be happy to set up a PR for them.

The rich types have proven to be really nice so far for migrating a numpy codebase, but as I'm sure you're aware [ulab can be built with with/without certain functions](https://github.com/v923z/micropython-ulab/blob/master/code/ulab.h). Perhaps to get the best of both worlds, the full ulab stubs could be built from the ulab repo and any unavailable functions (found by prodding the firmware on a board) could be commented out.

Josverl, 2020-12-08:
@CallumJHays 
I have added a list of ulab modules to the stubber list in the Dev/ulab branch.
could you test this version please ?

CallumJHays, 2020-12-10:
I've given it a go. The ulab stub folder is created but there are no stub files in it. Notably, the same thing happens for `umqtt` and `urllib`. Here's the output of the newly minified createstubs.py run on my esp32:

```python
>>> import createstubs
Clean/remove files in folder: /stubs/none
Clean/remove files in folder: /stubs/none/uasyncio
Stub module: uasyncio/__init__    to file: /stubs/none/uasyncio/__init__.py                        mem:98320
Stub module: uasyncio/core        to file: /stubs/none/uasyncio/core.py                            mem:93776
Stub module: uasyncio/event       to file: /stubs/none/uasyncio/event.py                           mem:93696
Stub module: uasyncio/funcs       to file: /stubs/none/uasyncio/funcs.py                           mem:93616
Stub module: uasyncio/lock        to file: /stubs/none/uasyncio/lock.py                            mem:93520
Stub module: uasyncio/stream      to file: /stubs/none/uasyncio/stream.py                          mem:93440
Stub module: ulab/approx          to file: /stubs/none/ulab/approx.py                              mem:93360
Stub module: ulab/compare         to file: /stubs/none/ulab/compare.py                             mem:93200
Stub module: ulab/fft             to file: /stubs/none/ulab/fft.py                                 mem:93136
Stub module: ulab/filter          to file: /stubs/none/ulab/filter.py                              mem:93216
Stub module: ulab/linalg          to file: /stubs/none/ulab/linalg.py                              mem:93072
Stub module: ulab/numerical       to file: /stubs/none/ulab/numerical.py                           mem:93072
Stub module: ulab/poly            to file: /stubs/none/ulab/poly.py                                mem:93040
Stub module: ulab/user            to file: /stubs/none/ulab/user.py                                mem:93104
Stub module: ulab/vector          to file: /stubs/none/ulab/vector.py                              mem:92976
Stub module: umqtt/robust         to file: /stubs/none/umqtt/robust.py                             mem:93072
Stub module: umqtt/simple         to file: /stubs/none/umqtt/simple.py                             mem:92848
Stub module: urllib/urequest      to file: /stubs/none/urllib/urequest.py                          mem:93040
Stub module: _thread              to file: /stubs/none/_thread.py                                  mem:92944
Stub module: ak8963               to file: /stubs/none/ak8963.py                                   mem:92992
Stub module: apa102               to file: /stubs/none/apa102.py                                   mem:92848
Stub module: apa106               to file: /stubs/none/apa106.py                                   mem:92880
Stub module: array                to file: /stubs/none/array.py                                    mem:92560
Stub module: binascii             to file: /stubs/none/binascii.py                                 mem:92464
Stub module: btree                to file: /stubs/none/btree.py                                    mem:92384
Stub module: builtins             to file: /stubs/none/builtins.py                                 mem:92288
Stub module: cmath                to file: /stubs/none/cmath.py                                    mem:92224
Stub module: collections          to file: /stubs/none/collections.py                              mem:92144
Stub module: crypto               to file: /stubs/none/crypto.py                                   mem:92064
Stub module: curl                 to file: /stubs/none/curl.py                                     mem:91824
Stub module: dht                  to file: /stubs/none/dht.py                                      mem:91824
Stub module: display              to file: /stubs/none/display.py                                  mem:91968
Stub module: ds18x20              to file: /stubs/none/ds18x20.py                                  mem:91872
Stub module: errno                to file: /stubs/none/errno.py                                    mem:91904
Stub module: esp                  to file: /stubs/none/esp.py                                      mem:91760
Stub module: esp32                to file: /stubs/none/esp32.py                                    mem:91680
Stub module: flashbdev            to file: /stubs/none/flashbdev.py                                mem:91600
Stub module: framebuf             to file: /stubs/none/framebuf.py                                 mem:91568
Stub module: freesans20           to file: /stubs/none/freesans20.py                               mem:91504
Stub module: functools            to file: /stubs/none/functools.py                                mem:91264
Stub module: gc                   to file: /stubs/none/gc.py                                       mem:91360
Stub module: gsm                  to file: /stubs/none/gsm.py                                      mem:91136
Stub module: hashlib              to file: /stubs/none/hashlib.py                                  mem:90928
Stub module: heapq                to file: /stubs/none/heapq.py                                    mem:91344
Stub module: inisetup             to file: /stubs/none/inisetup.py                                 mem:91264
Stub module: io                   to file: /stubs/none/io.py                                       mem:91104
Stub module: json                 to file: /stubs/none/json.py                                     mem:91008
[... cut ...]
builtins.py         heapq.py            random.py           ucryptolib.py       ure.py              zlib.py            
cmath.py            inisetup.py         re.py               uctypes.py          uselect.py          modules.json       
collections.py      io.py               select.py           uerrno.py           usocket.py         
/pyboard/stubs/none> ls ulab
/pyboard/stubs/none> ls umqtt/
/pyboard/stubs/none> ls uasyncio/
__init__.py core.py     event.py    funcs.py    lock.py     stream.py
```
Might be doing something wrong :(

Josverl, 2020-12-10:
What firmware are you using? If you can share a link to that version I can try flashing it on one of my devices and see what happens

CallumJHays, 2020-12-16:
Hi, sorry for the hiatus - haven't had my esp32 handy for the past week. My firmware version is v1.13-103 on the esp32 ([-103 as it fixes the enum bug required by the stubber](https://github.com/BradenM/micropy-cli/issues/167)). Here's what prints when I open up the repl:
```
MicroPython v1.13-103-gb137d064e on 2020-12-03; ESP32 module with ESP32
```

The output of `help('modules')` is:
```python
>>> help('modules')
__main__          framebuf          uasyncio/lock     upip_utarfile
_boot             gc                uasyncio/stream   urandom
_onewire          inisetup          ubinascii         ure
_thread           machine           ubluetooth        uselect
_uasyncio         math              ucollections      usocket
_webrepl          micropython       ucryptolib        ussl
apa106            neopixel          uctypes           ustruct
btree             network           uerrno            usys
builtins          ntptime           uhashlib          utime
cmath             onewire           uheapq            utimeq
dht               uarray            uio               uwebsocket
ds18x20           uasyncio/__init__ ujson             uzlib
esp               uasyncio/core     ulab              webrepl
esp32             uasyncio/event    uos               webrepl_setup
flashbdev         uasyncio/funcs    upip              websocket_helper
Plus any modules on the filesystem
```

Also, just to potentially save you some time, `ulab` does exist on the board whereas `umqtt` does not:
```python
>>> import ulab
>>> ulab.__version__
'1.1.0-2D-numpy'
>>> import umqtt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'umqtt'
```

I've uploaded the compiled firmware to a gist to try out, the file [is available here](https://gist.github.com/CallumJHays/86cff43126d093a84a999708fa7de126/raw/721c62b84ea081efa2bf3c78c224516106fc799b/firmware.bin). Thanks!

Josverl, 2020-12-16:
thanks, 

meanwhile I have figured out how to generate the ulab stubs from source.
I had not found a way to get the version yet , so seeing your .__version__ gives me a handle on that :-) 

they are in [the `dev` branch of the stubs repo](https://github.com/Josverl/micropython-stubs/tree/dev) 
these should provide rich completion, with the exception of the ulab.user module which i have exclude to prevent the empty stub module interfering.

CallumJHays, 2020-12-17:
Great to hear,

I've ran into something that may have something to do with the issue occurring with stub generation. For some reason submodules of ulab can't be imported directly:
```python
>>> import ulab
>>> ulab.linalg
<module 'linalg'>
>>> import ulab.linalg
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'ulab'
```

Just putting it out there in case it helps.

closed 2022-08-23 by Josverl