Josverl/micropython-stubber #64: Autodetection of modules in firmware

robertoetcheverryr, 2021-09-16:
Hi,
I'm working on getting my stubs prepared for a lv_micropython 1.17 for ESP32 firmware.
I tried micropy-cli first and then micropython-stubber since a forum post mentioned that micropy-cli did not collect all modules besides micropython's. Now, that forum post mentioned that I could edit createstubs.py to add the desired modules with the output of the help("modules") command.
Shouldn't it be possible to capture the output of the help("modules") command, parse it, discard the problematic modules, deduplicate the list against the originally hardcoded one and then continue with the stub generation? It won't catch all modules, but at least it would get modules added like lvgl (I am not certain if the term is "frozen" for those modules?)

I finally got my stubs after moving to a board with SPIRAM, since the lvgl stub file ended up clocking at around 600kb and that went over the paltry 82kb of RAM that my original board had available...
But in any case, I'd like to help automate that step.

Josverl, 2021-09-26:
Hi Robert,

I'd love to add that functionality to the stubber, and I did already try but unfortunately that is not possible. 
the `help("modules")` output cannot be re-directed to anything that can read from micropython 

the alternatives are: 
 **- use a PC to run `help("modules")`** 
    - read the response 
    - add / merge these to the modules to process ( not too difficult) 
    - upload the stubber.py (or list with additional modules) 
    - then run the stubber 
and that would need to work across OSses and connections.
micropy-cli does a number of these steps, but not the interrogation of any custom modules.

**Much simpler:** 
its not to difficult to read a list of additional modules from an `extra.txt`  file and add / prepend these to the list of modules to process .
you will still need to add the extra module to that file ( copy / paste using you editor of choice) 

**Simplest:**
I can add `lvgl` to the list of modules to be stubbed
if it is a large module , then that may cause some problems with storage.

Lastly, if you have stubs of a module , and these are useful to others , please do share them and I will add them to the https://github.com/Josverl/micropython-stubs repo

Josverl, 2021-09-26:
Also see : https://github.com/Josverl/micropython-stubs/issues/97

closed 2024-11-20 by Josverl