Josverl/micropython-stubber #308: ValueError: not enough values to unpack (expected 4, got 1)

rpetersnil, 2023-01-18:
When executing 'micropy stubs create -v /dev/cu.usbserial-110', I get the following error (my understandign this is from this app, not micropy, please correct if wrong): 

MicroPy  Executing stubber on pyboard...
MicroPy  Pyboard  Writing /createstubs.py: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████| [8.14k/8.14k @ 1.23kB/s]
MicroPy  ✔ Done!
MicroPy  Copying stubs...
Traceback (most recent call last):
  File "/opt/homebrew/bin/micropy", line 8, in <module>
    sys.exit(cli())
  File "/opt/homebrew/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/opt/homebrew/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/opt/homebrew/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/opt/homebrew/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/opt/homebrew/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/opt/homebrew/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/opt/homebrew/lib/python3.10/site-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/opt/homebrew/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/opt/homebrew/lib/python3.10/site-packages/micropy/cli.py", line 276, in create
    return mpy.create_stubs(port, verbose=verbose)
  File "/opt/homebrew/lib/python3.10/site-packages/micropy/main.py", line 128, in create_stubs
    pyb.copy_from(DevicePath("/stubs"), tmpdir)
  File "/opt/homebrew/lib/python3.10/site-packages/micropy/pyd/pydevice.py", line 41, in copy_from
    return self.pydevice.copy_dir(
  File "/opt/homebrew/lib/python3.10/site-packages/micropy/pyd/backend_upydevice.py", line 137, in copy_dir
    for file_path in self.iter_files(source_path):
  File "/opt/homebrew/lib/python3.10/site-packages/micropy/pyd/backend_upydevice.py", line 127, in iter_files
    for name, type_, _, _ in results:
ValueError: not enough values to unpack (expected 4, got 1)

Josverl, 2023-01-29:
Hi Peter 

I think this is more of a micropy-cli issue as I do not see any of the micropython-stubber filenames popping up in the stactrace 
perhaps @BradenM can take a look

BradenM, 2023-01-30:
@rpetersnil 
Hi,

@Josverl is correct -- this looks to be an issue with micropy-cli, not stubber.

If you wouldn't mind creating an issue over at [BradenM/micropy-cli](https://github.com/BradenM/micropy-cli) and attaching your micropy log file (see the issue template for details), I can gladly take a more in-depth look. Thanks.

closed 2023-02-05 by Josverl