Josverl/micropython-stubber #501: BTLE low-level constants defined as 'Incomplete'

Josverl, 2024-01-26:
stub: `publish\micropython-v1_21_0-rp2-rpi_pico_w-stubs\aioble\client.pyi`

The types should be 'int', not `incomplete` 

```py
# .Py source 
_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)

_IRQ_PERIPHERAL_CONNECT = const(7)
_IRQ_PERIPHERAL_DISCONNECT = const(8)

_ADV_IND = const(0)
_ADV_DIRECT_IND = const(1)
_ADV_SCAN_IND = const(2)
_ADV_NONCONN_IND = const(3)
_SCAN_RSP = const(4)

_ADV_TYPE_FLAGS = const(0x01)
_ADV_TYPE_NAME = const(0x09)
_ADV_TYPE_SHORT_NAME = const(0x08)
_ADV_TYPE_UUID16_INCOMPLETE = const(0x2)
_ADV_TYPE_UUID16_COMPLETE = const(0x3)
_ADV_TYPE_UUID32_INCOMPLETE = const(0x4)
_ADV_TYPE_UUID32_COMPLETE = const(0x5)
_ADV_TYPE_UUID128_INCOMPLETE = const(0x6)
_ADV_TYPE_UUID128_COMPLETE = const(0x7)
_ADV_TYPE_APPEARANCE = const(0x19)
_ADV_TYPE_MANUFACTURER = const(0xFF)

# .pyi stub

_IRQ_SCAN_RESULT: Incomplete
_IRQ_SCAN_DONE: Incomplete
_IRQ_PERIPHERAL_CONNECT: Incomplete
_IRQ_PERIPHERAL_DISCONNECT: Incomplete
_ADV_IND: Incomplete
_ADV_DIRECT_IND: Incomplete
_ADV_SCAN_IND: Incomplete
_ADV_NONCONN_IND: Incomplete
_SCAN_RSP: Incomplete
_ADV_TYPE_FLAGS: Incomplete
_ADV_TYPE_NAME: Incomplete
_ADV_TYPE_SHORT_NAME: Incomplete
_ADV_TYPE_UUID16_INCOMPLETE: Incomplete
_ADV_TYPE_UUID16_COMPLETE: Incomplete
_ADV_TYPE_UUID32_INCOMPLETE: Incomplete
_ADV_TYPE_UUID32_COMPLETE: Incomplete
_ADV_TYPE_UUID128_INCOMPLETE: Incomplete
_ADV_TYPE_UUID128_COMPLETE: Incomplete
_ADV_TYPE_APPEARANCE: Incomplete
_ADV_TYPE_MANUFACTURER: Incomplete
_active_scanner: Incomplete
_connecting: Incomplete
```

Reported by @brianreinhold in : https://github.com/orgs/micropython/discussions/13527

brianreinhold, 2024-01-26:
Maybe the solution is to install the source (at least on my PC as the PICO is being accessed remotely) which would provide pylance with the needed information. That seems like a good solution if the actual values exist in the source. That would really help me understand the code and how to use it.

That aside, if Incomplete were changed to int, would that cause a problem?

Josverl, 2024-01-27:
I agree that `int` would be better, it's just that the tools that turn the source into stubs do not seem to understand the micropython `const` directive.

And for understanding, the source is better than the stubs, as the stubs never have the implementation

Josverl, 2024-02-06:
@brianreinhold, 
Thanks for the report. 
I updated the stubber tool to allow the types of almost all consts to be preserved during the stubbing process.
That in turn improves the typing across all port and frozen modules. 

Updates have been published to PyPI as micropython-rp2-rpi_pico_w-stubs  v1.22.1.post2
If you have modified your local stubs, make sure they do not get overwritten by an automatic update.

closed 2024-02-06 by Josverl

brianreinhold, 2024-02-06:
That is really helpful. I have replaced the stubs with the source in the .vscode directory (aioble only) such that pylance references the source. The source still has a lot of `None`s where a typing would be much more helpful. As I figure out what they are, I update them.

Josverl, 2024-02-07:
Large parts are sparsely documented. 
If you have a set that you think is useful I'd be happy to take a copy (in a gist or something) and try to find a way to integrate them.

commit ac30b808 (2024-02-14): get-frozen: preprocess stubs to improve typing for `const`s
fixes #501
PR #510 (2024-02-14): Fix BTLE low-level constants type in aioble/client.pyi
This pull request fixes the issue with the BTLE low-level constants in the `aioble/client.pyi` stub file. The constants were defined as `Incomplete` instead of `int`. 

This PR removes all single line `const()`  from the to-be frozen scripts so that mypy.stubgen does not error out on them.
The replacement is done using a regex - that gets 90%+ of the constants. 
Full coverage is possible by replacing the regex with a CSTLib codemod - likely a bit slower.

This generates the proper types for the majority of module level CONSTANTS.

Fixes #5 [... cut]