Josverl/micropython-stubber #550: MPFlash unnecessarily verbose error message

andrewleech, 2024-04-30:
**Provide your feedback**
My first try was with a board connected that's not yet supported by micropython (it's running code from support PR)

The reported error is correct, but the stack trace provided is very long and I don't know that it really adds value here

```
anl@STEP: ~/micropython $ mpflash list
Traceback (most recent call last):
  File "/home/anl/.local/bin/mpflash", line 8, in <module>
    sys.exit(mpflash())
  File "/home/anl/.local/lib/python3.10/site-packages/mpflash/cli_main.py", line 19, in mpflash
    result = cli(standalone_mode=False)
  File "/home/anl/.local/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/anl/.local/lib/python3.10/site-packages/rich_click/rich_command.py", line 126, in main
    rv = self.invoke(ctx)
  File "/home/anl/.local/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/anl/.local/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/anl/.local/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/anl/.local/lib/python3.10/site-packages/mpflash/cli_list.py", line 35, in cli_list_mcus
    conn_mcus = list_mcus()
  File "/home/anl/.local/lib/python3.10/site-packages/mpflash/list.py", line 39, in list_mcus
    mcu.get_mcu_info()
  File "/home/anl/.local/lib/python3.10/site-packages/tenacity/__init__.py", line 289, in wrapped_f
    return self(f, *args, **kw)
  File "/home/anl/.local/lib/python3.10/site-packages/tenacity/__init__.py", line 379, in __call__    do = self.iter(retry_state=retry_state)
  File "/home/anl/.local/lib/python3.10/site-packages/tenacity/__init__.py", line 325, in iter
    raise retry_exc.reraise()
  File "/home/anl/.local/lib/python3.10/site-packages/tenacity/__init__.py", line 158, in reraise
    raise self.last_attempt.result()
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 451, in result
    return self.__get_result()
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
    raise self._exception
  File "/home/anl/.local/lib/python3.10/site-packages/tenacity/__init__.py", line 382, in __call__    result = fn(*args, **kwargs)
  File "/home/anl/.local/lib/python3.10/site-packages/mpflash/mpremoteboard/__init__.py", line 119, in get_mcu_info
    if board_name := find_board_id(descr, short_descr):
  File "/home/anl/.local/lib/python3.10/site-packages/mpflash/mpboard_id/board_id.py", line 22, in find_board_id
    boards = find_board_by_description(
  File "/home/anl/.local/lib/python3.10/site-packages/mpflash/mpboard_id/board_id.py", line 55, in find_board_by_description
    raise MPFlashError(f"No board info found for
description {descr}")
mpflash.errors.MPFlashError: No board info found
for description ESP32C6 module with ESP32C6
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run `mpflash list` with unsupported board connected.

**Expected behavior**
In this case I think just the final error message should be shown without the stack trace. Maybe add a verbose comment love flag to allow showing the trace of needed.

**Hardware Info**
Host
 - Linux (WSL2 on windows running Ubuntu)

MCU
 - Family: micropython 
 - Version build from PR
 - esp32c6 
 - usb serial / ACM connection

Josverl, 2024-05-01:
The error is now ( v0.7.2) caught during `list` processing and the board_id is reported as UNKOWN_BOARD in the list output.
 
I expect that the in some cases download and flash may still throw a similar Exception. I want to think a bit longer on how this case should be handled for those usecases, and a few more tests need to be added

Josverl, 2024-05-06:
confirmed working

closed 2024-05-06 by Josverl