Josverl/micropython-stubber #551: MPFlash list output doesn't fit a standard terminal

scruss, 2024-05-01:
**Provide your feedback**
list output, though formatted, doesn't fit in a standard 80-column terminal

**To Reproduce**
Steps to reproduce the behaviour:
1. Open a terminal with Linux system defaults (`xterm`, if it's installed, defaults to 80 columns
2. Issue `mpflash list`
3. See disappointing truncated results

```
mpflash list
Error: Failed to get mcu_info for /dev/ttyS0
Error: Failed to get mcu_info for /dev/ttyUSB0

                                               Connected boards                 
┏━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━
┃ Serial  ┃Family        ┃Port         ┃Board                               ┃CPU
┡━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━
│ ttyACM0 │circuitpython │Atmel SAMD21 │UNKNOWN_BOARD                       │sam
│         │              │             │Adafruit QT Py M0 with samd21e18    │   
│ ttyACM1 │micropython   │rp2          │RPI_PICO_W                          │RP2
│         │              │             │Raspberry Pi Pico W with RP2040     │   
│ ttyACM2 │micropython   │mimxrt       │ADAFRUIT_METRO_M7                   │MIM
│         │              │             │Adafruit Metro M7 with              │   
│         │              │             │MIMXRT1011DAE5A                     │   
│ ttyS0   │unknown       │             │                                    │   
│ ttyUSB0 │unknown       │             │                                    │   
└─────────┴──────────────┴─────────────┴────────────────────────────────────┴───

```

**Expected behaviour**
Be able to see all columnar output in the terminal. It looks like it's hard-coded to use something like 110 columns. The scanning code respects terminal width. This is an accessibility issue for me.

**Hardware Info**
Host
 - Ubuntu 22.04.4 LTS, x86_64

MCU
`mpflash list --json`  output:
```json
[
    {
        "serialport": "/dev/ttyACM0",
        "firmware": {},
        "connected": true,
        "path": null,
        "family": "circuitpython",
        "description": "Adafruit QT Py M0 with samd21e18",
        "version": "7.2.5",
        "port": "Atmel SAMD21",
        "board": "UNKNOWN_BOARD",
        "cpu": "samd21e18",
        "arch": "",
        "mpy": "v5.2",
        "build": ""
    },
    {
        "serialport": "/dev/ttyACM1",
        "firmware": {},
        "connected": true,
        "path": null,
        "family": "micropython",
        "description": "Raspberry Pi Pico W with RP2040",
        "version": "1.21.0",
        "port": "rp2",
        "board": "RPI_PICO_W",
        "cpu": "RP2040",
        "arch": "armv6m",
        "mpy": "v6.1",
        "build": ""
    },
    {
        "serialport": "/dev/ttyACM2",
        "firmware": {},
        "connected": true,
        "path": null,
        "family": "micropython",
        "description": "Adafruit Metro M7 with MIMXRT1011DAE5A",
        "version": "1.21.0",
        "port": "mimxrt",
        "board": "ADAFRUIT_METRO_M7",
        "cpu": "MIMXRT1011DAE5A",
        "arch": "armv7emsp",
        "mpy": "v6.1",
        "build": ""
    },
    {
        "serialport": "/dev/ttyS0",
        "firmware": {},
        "connected": false,
        "path": null,
        "family": "unknown",
        "description": "",
        "version": "",
        "port": "",
        "board": "",
        "cpu": "",
        "arch": "",
        "mpy": "",
        "build": ""
    },
    {
        "serialport": "/dev/ttyUSB0",
        "firmware": {},
        "connected": false,
        "path": null,
        "family": "unknown",
        "description": "",
        "version": "",
        "port": "",
        "board": "",
        "cpu": "",
        "arch": "",
        "mpy": "",
        "build": ""
    }
]
```

Josverl, 2024-05-01:
Thanks for the feedback. 
I've been trying to get the relevant info on screen with the rich Table interface but it has its own quirks when setting total and columns width.
Ive indeed taken the simple approach to set the width to 110.

As this is an accessibility issue for you I will address this, but will likely need to drop, or abbreviate/ or fold  some columns to fit to 80 characters.

can you help by letting me know which of the columns are most relevant to you ?

1.  table.add_column("Serial", overflow="fold")
2.  table.add_column("Family")
3.  table.add_column("Port")
4.  table.add_column("Board", overflow="fold")
5.  # table.add_column("Variant") # TODO: add variant
6.  table.add_column("CPU")
7.  table.add_column("Version")
8.  table.add_column("build", justify="right")

Josverl, 2024-05-01:
I've been doing some testing with adjusting depending on the detected terminal width, would like your feedback on these 

- removed the static width setting
- the Build column is only shown if there is data to be shown.
- removed the default padding 

if the console.width < 100 : 
- use abbriviations for long headers 
- truncate the familiy to max 4 chars 
- set the version max_width to 8 
- use abbriveations for the family

this  results in : 
### 80 column list 
```txt
┏━━━━━┳━━━━┳━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━┳━━━┓
┃Ser. ┃Fam.┃Port    ┃Board                            ┃CPU       ┃Version ┃Bld┃  
┡━━━━━╇━━━━╇━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━╇━━━┩  
│COM27│cpy │nRF52840│UNKNOWN_BOARD                    │nRF52840  │v8.2.10 │   │  
│     │    │        │nice!nano with nRF52840          │          │        │   │  
│COM8 │upy │samd    │SEEED_WIO_TERMINAL               │SAMD51P19A│v1.23.0-│341│  
│     │    │        │Wio Terminal D51R with SAMD51P19A│          │preview │   │  
└─────┴────┴────────┴─────────────────────────────────┴──────────┴────────┴───┘
```

On a > 100 char terminal  ( note that the table is still only 99 chars wide
this is driven by the length of the descriptions.
```txt
                                         Connected boards
┏━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━━━━━━━━┳━━━━━┓
┃Serial┃Family       ┃Port    ┃Board                            ┃CPU       ┃Version        ┃Build┃
┡━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━━━━━━━━╇━━━━━┩
│COM27 │circuitpython│nRF52840│UNKNOWN_BOARD                    │nRF52840  │v8.2.10        │     │
│      │             │        │nice!nano with nRF52840          │          │               │     │
│COM8  │micropython  │samd    │SEEED_WIO_TERMINAL               │SAMD51P19A│v1.23.0-preview│  341│
│      │             │        │Wio Terminal D51R with SAMD51P19A│          │               │     │
└──────┴─────────────┴────────┴─────────────────────────────────┴──────────┴───────────────┴─────┘ 
```

if this works - then there is no need for now to drop any columns , 
and if I need to add Variant later, then it can probably be added in the Board column

scruss, 2024-05-02:
> Can you help by letting me know which of the columns are most relevant to you ?
> 

Essential:
>     1. table.add_column("Serial", overflow="fold")
>     4. table.add_column("Board", overflow="fold") # Board _ID and description on separate lines
>     7. table.add_column("Version")

Useful (especially in truncated form):
>     2. table.add_column("Family")

Duplicated/implied in the Board column:
>     3. table.add_column("Port")
>     6. table.add_column("CPU")

Can't see a use for (and could quickly find out from the REPL):
>     8. table.add_column("build", justify="right")
>     5. table.add_column("Variant") # TODO: not a current column

> Also - is there anything I should take into account wrt to the colors ?

I can't comment on how they work for other people, but they're good and contrasty here. I'm not a screen reader user, but I know a lot of screen readers get very shouty if they see bold or italics

Josverl, 2024-05-02:
Thanks,
Ive made the Port and CPU columns drop out on terminals <100 characters. 
I use the Build to keep track of different preview versions - so Ill keep that , but its displayed only if there is any info in it.

The result is :
``` txt
mpflash list

                         Connected boards                                      
┏━━━━━┳━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━┳━━━┓
┃Ser. ┃Fam.┃Board                            ┃Version        ┃Bld┃
┡━━━━━╇━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━╇━━━┩
│COM27│cpy │UNKNOWN_BOARD                    │v8.2.10        │   │
│     │    │nice!nano with nRF52840          │               │   │
│COM8 │upy │SEEED_WIO_TERMINAL               │v1.23.0-preview│341│
│     │    │Wio Terminal D51R with SAMD51P19A│               │   │
└─────┴────┴─────────────────────────────────┴───────────────┴───┘
```

Based on you comment about shouty screenreaders I did a quick test with Windows narrator and I found that the progress bar updates were leading to a  highly annoying an repetitive narration.
I could not hear a difference between Bold/Italics - so I've left these as is for now.
I lowered the update frequency on that to find - what I think is a balance between visual and narrative updates.

PR #552 (2024-05-03): mpflash: Improve list display on 80 column terminals

- removed the static width setting
- the Build column is only shown if there is data to be shown.
- removed the default padding
- serial ports on unix are stripped of the leading /dev/

if the console.width < 100 :

- use abbriviations for long headers
- truncate the familiy to max 4 chars
- set the version max_width to 8
- use abbreviations for the family

Fixes #551
closed 2024-05-03 by Josverl