Josverl/micropython-stubber #553: mpflash: rp2 download fails

Josverl, 2024-05-01:
I like it!

It doesn't seem to be able to download Pico W firmware, though:
```
mpflash download --board RPI_PICO_W
20:55:35 | ℹ️ Found 0 relevant unique firmwares
20:55:35 | ℹ️ Downloaded 0 firmwares, skipped 0 existing files.
```

_Originally posted by @scruss in https://github.com/micropython/micropython/discussions/14395#discussioncomment-9280971_

Josverl, 2024-05-06:
@scruss,
can you retest with v0.7.5 or newer ?
I fixes a lookup bug for the BOARD_IDs that affected both ESPxx and RP2 boards.

> mpflash download --board RPI_PICO_W --force  
23:54:38 | ℹ️ Found 1 relevant unique firmwares                                                                                                                                                 
23:54:38 | ℹ️ Downloading https://micropython.org/resources/firmware/RPI_PICO_W-20240222-v1.22.2.uf2
23:54:38 | ℹ️          to C:\Users\josverl\Downloads\firmware\rp2\RPI_PICO_W-v1.22.2.uf2
23:54:39 | ℹ️ Downloaded 1 firmwares, skipped 0 existing files.

scruss, 2024-05-08:
Download works now, thanks:

```
mpflash download --board RPI_PICO_W --force
13:16:15 | ℹ️ Found 1 relevant unique firmwares
13:16:15 | ℹ️ Downloading 
https://micropython.org/resources/firmware/RPI_PICO_W-20240222-v1.22.2.uf2
13:16:15 | ℹ️          to 
/home/scruss/Downloads/firmware/rp2/RPI_PICO_W-v1.22.2.uf2
13:16:16 | ℹ️ Downloaded 1 firmwares, skipped 0 existing files.
```

(a separate issue is that *flash* doesn't: I have serial ports other than MicroPython devices, and I'm fairly sure the DEC PDP-8 minicomputer hanging off /dev/ttyS0 is not a supported board)

Josverl, 2024-05-08:
An actual PDP-8 😁In high school computer science I was given access to a PDP-11 using scratch cards an a batch reader. The  PDP-8 next to it onlly had a TTY and iirc only had FORTH and it's ring core memory was a bit noisy. 

To ignore a serialport you can use `mpflash - - ignore /dev/someport` or set an en variable MPFLASH_IGNORE.

closed 2024-05-09 by Josverl