Josverl/micropython-stubber #562: mpflash: Improve HTTP error handling

Josverl, 2024-05-07:
[ ] Improve error handling 
 [ ] check if multiple `--ignores` are possible

```
C:\Users\Owner>mpflash list
Error: Failed to get mcu_info for COM1
Error: Failed to get mcu_info for COM4
Error: Failed to get mcu_info for COM5

                              Connected boards
┏━━━━━━┳━━━━━━━━━━━┳━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━┳━━━━━━━┓
┃Serial┃Family     ┃Port ┃Board                         ┃CPU        ┃Version┃
┡━━━━━━╇━━━━━━━━━━━╇━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━╇━━━━━━━┩
│COM1  │unknown    │     │                              │           │       │
│COM4  │unknown    │     │                              │           │       │
│COM5  │unknown    │     │                              │           │       │
│COM9  │micropython│stm32│NUCLEO_L432KC                 │STM32L432KC│v1.22.2│
│      │           │     │NUCLEO-L432KC with STM32L432KC│           │       │
└──────┴───────────┴─────┴──────────────────────────────┴───────────┴───────┘

C:\Users\Owner>mpflash download
Error: Failed to get mcu_info for COM1
Error: Failed to get mcu_info for COM4
Error: Failed to get mcu_info for COM5

Traceback (most recent call last):
  File "E:\Python312\Lib\site-packages\urllib3\connection.py", line 198, in _new_conn
    sock = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\urllib3\util\connection.py", line 85, in create_connection
    raise err
  File "E:\Python312\Lib\site-packages\urllib3\util\connection.py", line 73, in create_connection
    sock.connect(sa)
TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "E:\Python312\Lib\site-packages\urllib3\connectionpool.py", line 793, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\urllib3\connectionpool.py", line 491, in _make_request
    raise new_e
  File "E:\Python312\Lib\site-packages\urllib3\connectionpool.py", line 467, in _make_request
    self._validate_conn(conn)
  File "E:\Python312\Lib\site-packages\urllib3\connectionpool.py", line 1099, in _validate_conn
    conn.connect()
  File "E:\Python312\Lib\site-packages\urllib3\connection.py", line 616, in connect
    self.sock = sock = self._new_conn()
                       ^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\urllib3\connection.py", line 207, in _new_conn
    raise ConnectTimeoutError(
urllib3.exceptions.ConnectTimeoutError: (<urllib3.connection.HTTPSConnection object at 0x00000204A123AA50>, 'Connection to micropython.org timed out. (connect timeout=None)')

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "E:\Python312\Lib\site-packages\requests\adapters.py", line 486, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\urllib3\connectionpool.py", line 847, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\urllib3\util\retry.py", line 515, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='micropython.org', port=443): Max retries exceeded with url: /download/?port=stm32 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x00000204A123AA50>, 'Connection to micropython.org timed out. (connect timeout=None)'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "E:\Python312\Scripts\mpflash.exe\__main__.py", line 7, in <module>
  File "E:\Python312\Lib\site-packages\mpflash\cli_main.py", line 20, in mpflash
    result = cli(standalone_mode=False)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\rich_click\rich_command.py", line 360, in __call__
    return super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\rich_click\rich_command.py", line 152, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\mpflash\cli_download.py", line 91, in cli_download
    download(
  File "E:\Python312\Lib\site-packages\mpflash\download.py", line 284, in download
    return download_firmwares(destination, ports, boards, versions, force=force, clean=clean)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\mpflash\download.py", line 177, in download_firmwares
    unique_boards = get_firmware_list(ports, boards, versions, clean)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\mpflash\download.py", line 228, in get_firmware_list
    board_urls = sorted(get_boards(ports, boards, clean), key=key_fw_ver_pre_ext_bld)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\mpflash\download.py", line 117, in get_boards
    _urls = get_board_urls(download_page_url)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\mpflash\download.py", line 55, in get_board_urls
    downloads_html = get_page(page_url)
                     ^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\mpflash\download.py", line 42, in get_page
    response = requests.get(page_url)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\requests\api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\requests\api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\requests\sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\requests\sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Python312\Lib\site-packages\requests\adapters.py", line 507, in send
    raise ConnectTimeout(e, request=request)
requests.exceptions.ConnectTimeout: HTTPSConnectionPool(host='micropython.org', port=443): Max retries exceeded with url: /download/?port=stm32 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPSConnection object at 0x00000204A123AA50>, 'Connection to micropython.org timed out. (connect timeout=None)'))
```

_Originally posted by @garryp4 in https://github.com/micropython/micropython/discussions/14395#discussioncomment-9343357_

Josverl, 2024-05-07:
@garryp4,

Can you open a python repl and try the below commands ?
```py
import requests
page = requests.get("https://micropython.org/download/?port=stm32")
print(page)
```
you should get an output similar to : 
``` log 
> python
Python 3.11.7 (tags/v3.11.7:fa7a6f2, Dec  4 2023, 19:24:49) [MSC v.1937 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import requests
>>> page = requests.get("https://micropython.org/download/?port=stm32")
>>> print(page)
<Response [200]>
>>>
```

garryp4, 2024-05-09:
Josverl:

mpflash.exe is loaded on my PC. When I look back at what I sent you, for some reason I tried to do another download. Not sure why I did that. If I do mpflash list, it works fine.

Josverl, 2024-05-12:
Does the download now succeed?

I still plan to catch and handle such errors, and it would help to understand what the problem was

closed 2024-06-02 by Josverl