Josverl/micropython-stubber #635: mpflash: MacOS ESP-Prog detection hangs

lukaskremla, 2024-12-23:
**Describe the bug**
This bug might not be bound to just MacOS, it might also happen on windows, but I haven't tried it there.

I'm working with an ESP32 S2 chip and I'm communicating with it through JTAG via the [ESP-Prog](https://docs.espressif.com/projects/esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html).

<details>
  <summary>Mpflash fails to identify the ESP32 S2 chip, which I expected, however it hangs on this for a few seconds.</summary>
  

https://github.com/user-attachments/assets/7fd76d52-e4ba-48c5-9c6b-b0bd384550c9

  </details>

Here is the output of:

```Python
import jsons
import serial.tools.list_ports as list_ports

comports = jsons.dumps(list_ports.comports())
print(comports)
```

<details>
  <summary>Formatted json</summary>
  

```json
[
  {
    "description": "n/a",
    "device": "/dev/cu.debug-console",
    "hwid": "n/a",
    "interface": null,
    "location": null,
    "manufacturer": null,
    "name": "cu.debug-console",
    "pid": null,
    "product": null,
    "serial_number": null,
    "vid": null
  },
  {
    "description": "n/a",
    "device": "/dev/cu.Hermes",
    "hwid": "n/a",
    "interface": null,
    "location": null,
    "manufacturer": null,
    "name": "cu.Hermes",
    "pid": null,
    "product": null,
    "serial_number": null,
    "vid": null
  },
  {
    "description": "n/a",
    "device": "/dev/cu.Bluetooth-Incoming-Port",
    "hwid": "n/a",
    "interface": null,
    "location": null,
    "manufacturer": null,
    "name": "cu.Bluetooth-Incoming-Port",
    "pid": null,
    "product": null,
    "serial_number": null,
    "vid": null
  },
  {
    "description": "Espressif Device",
    "device": "/dev/cu.usbmodem2101",
    "hwid": "USB VID:PID=303A:4001 SER=983daee912f0cb3f LOCATION=2-1",
    "interface": null,
    "location": "2-1",
    "manufacturer": "Espressif Systems",
    "name": "cu.usbmodem2101",
    "pid": 16385,
    "product": "Espressif Device",
    "serial_number": "983daee912f0cb3f",
    "vid": 12346
  },
  {
    "description": "Dual RS232-HS",
    "device": "/dev/cu.usbserial-1101",
    "hwid": "USB VID:PID=0403:6010 LOCATION=1-1",
    "interface": null,
    "location": "1-1",
    "manufacturer": "FTDI",
    "name": "cu.usbserial-1101",
    "pid": 24592,
    "product": "Dual RS232-HS",
    "serial_number": null,
    "vid": 1027
  },
  {
    "description": "Dual RS232-HS",
    "device": "/dev/cu.usbserial-1100",
    "hwid": "USB VID:PID=0403:6010 LOCATION=1-1",
    "interface": null,
    "location": "1-1",
    "manufacturer": "FTDI",
    "name": "cu.usbserial-1100",
    "pid": 24592,
    "product": "Dual RS232-HS",
    "serial_number": null,
    "vid": 1027
  }
]
```

  </details>
  
The submodem2101 was identified correctly as an ESP32 S3 and isn't connected via JTAG.
  
Tested through iterm2 and PyCharm with mpflash version 1.24.2. post1 and Python 3.10.11.
 
**Hardware information**
- MacBook M2 Pro
- MacOS Sequoia 15.2
- (This might concern all macOS computers)

Josverl, 2025-02-20:
I finally got around to digging into this.
I have a ESP-Progboard setup and I can identify the ports pasted on the `USB VID:PID=0403:6010` and skip them similar as skipping the bluetooth ports.

[image]

Or is there a way that the UART/Programmer can actually be used to connect to the ESPxxx board and update firmware from esptool ?
so something like 
```mermaid
graph LR;
    Host[esptool] --[com1|com2]--> B[ESP-Prog];
    B --[jtag | programmer] --> C[ESPxxx];
```

lukaskremla, 2025-02-20:
I'm using the ESP prog to flash firmware with esptool onto an ESP32 S2 using the Prog 2.54 / Program UART connector just fine. So it would be nice if mpflash was capable of doing it as well.

Josverl, 2025-02-20:
Can you share a link or description of the setup and wiring so I can replicate?