Josverl/micropython-stubber #646: I want to know the difference between these two warehouses

Tinyu-Zhao, 2025-01-16:
https://github.com/Josverl/micropython-stubber
https://github.com/Josverl/micropython-stubs

Their names and contents are so similar that I looked at the description and couldn't find the difference between them; if I just want to use the stub file .pyi in vscode, which one is better to choose.

Tinyu-Zhao, 2025-01-16:
Stubber is used to create stubs, 😂 similar readthedocs misled me, thank you for your contribution, but I want to know why not synchronize from the latest mpy version, which is now much behind v1.4.4.

Josverl, 2025-01-19:
Indeed, I split the repo for the stubber tool from the repo for the generated stubs in order to avoid problems with complex merges and cherry picks.

I am currently working on adding support for v1.24. But this is taking some more time than expected because I'm also adding a lot more functionality with regards to. typing for all/most parameters of all the methods.

I expect to have them available in the coming weeks. If you're interested in testing or validating them, please let me know which port and board that you are interested in, and then I can see if I can get you a link to do the installation and validation.

Tinyu-Zhao, 2025-01-20:
> Indeed, I split the repo for the stubber tool from the repo for the generated stubs in order to avoid problems with complex merges and cherry picks.
>
> I am currently working on adding support for [...]

I've been using esp32 and I'd love to try out the pre-release boards.

I think we can try to prioritize adapting the latest mpy stubs, and then work backwards from the latest version, instead of always playing catch-up at the end!

Josverl, 2025-01-20:
Hi,
Creating the stubs for a version is still a significant amount of manual work, even if it's largely automated.

In addition I'm making structural changes to improve the quality, by adding much more details regarding the parameters to methods and functions.

This is taking more time than I anticipated, but I'm well on the way.
The v1.23.c stubs are largely correct for v1.24.x as well 

If your interested in testing a new stub package, let me know the port and bo

Tinyu-Zhao, 2025-01-21:
> The v1.23.x stubs are largely correct for v1.24.x as well, But for v1.24 I would have needed to undo a lot of the changes that were already in place.
> 
> As I said , the challenges of merging type info and developing proved more complex than expected for this devteam of one.

Sorry, I saw that the document version of mpy said v1.4.4. I mistakenly thought that the version number of the document was synchronized with the version number of mpy version (just like the official document of ESP-IDF). Thank you for your contribution.

[image]

[image]

closed 2025-01-21 by Tinyu-Zhao