Josverl/micropython-stubber #663: error trying to run "stubber get-mcu-stubs --serial com3" on windows 11

jonhp, 2025-05-04:
```
13:14:29|INFO    |runner               - All modules have been processed, Finalizing report
13:14:29|INFO    |runner               - Path: /remote/stubs/micropython-v1_24_1-esp32
13:14:29|INFO    |runner               - Local directory C:\Users\jon\AppData\Local\Temp\board_stubberfvtgohx4 is mounted at /remote
13:14:31|INFO    |post                 - Running autoflake on: C:\Users\jon\AppData\Local\Temp\board_stubberfvtgohx4\stubs\micropython-v1_24_1-esp32
13:14:31|SUCCESS |mcu_stubber          - Stubs generated for esp32-
13:14:31|SUCCESS |mcu_stubber          - Stubs copied to repos\micropython-stubs\stubs\micropython-v1_24_1-esp32
13:14:31|INFO    |mcu_stubber          - Merging stubs with docstubs : {'variant': '', 'build': '', 'arch': 'xtensawin', 'port': 'esp32', 'board': '', 'board_id': '', 'mpy': 'v6.3', 'ver': '1.24.1', 'family': 'micropython', 'cpu': 'ESP32S3', 'version': '1.24.1'}
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Python313\Scripts\stubber.exe\__main__.py", line 7, in <module>
    sys.exit(stubber_cli())
             ~~~~~~~~~~~^^
  File "C:\Python313\Lib\site-packages\rich_click\rich_command.py", line 378, in __call__
    return super().__call__(*args, **kwargs)
           ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Python313\Lib\site-packages\click\core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Python313\Lib\site-packages\rich_click\rich_command.py", line 166, in main
    rv = self.invoke(ctx)
  File "C:\Python313\Lib\site-packages\click\core.py", line 1728, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
              ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "C:\Python313\Lib\site-packages\click\core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python313\Lib\site-packages\click\core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "C:\Python313\Lib\site-packages\stubber\commands\get_mcu_cmd.py", line 94, in cli_create_mcu_stubs
    stub_connected_mcus(
    ~~~~~~~~~~~~~~~~~~~^
        variant=variant,
        ^^^^^^^^^^^^^^^^
    ...<4 lines>...
        bluetooth=bluetooth,
        ^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "C:\Python313\Lib\site-packages\stubber\bulk\mcu_stubber.py", line 391, in stub_connected_mcus
    merged = merge_all_docstubs(
        versions=board.firmware["version"],
    ...<3 lines>...
        mpy_path=CONFIG.mpy_path,
    )
TypeError: merge_all_docstubs() got an unexpected keyword argument 'mpy_path'

```

Josverl, 2025-05-04:
There was an extra parameter left-over from an earlier change that I fixed as part of a future version, which is not yet released to PyPI

I have fixed this , tested it, and published v1.24.4.post1  to PyPI.

Josverl, 2025-05-04:
please reinstall / update to verify it solves the issue reported.

jonhp, 2025-05-10:
ran a long time and then:

08:17:56|INFO    |runner               - All modules have been processed, Finalizing report
08:17:56|INFO    |runner               - Path: /remote/stubs/micropython-v1_25_0-esp32-ESP32_GENERIC_S3
08:17:56|INFO    |runner               - Local directory C:\Users\jon\AppData\Local\Temp\board_stubberpxc9nckz is mounted at /remote
08:17:58|INFO    |post                 - Running autoflake on: C:\Users\jon\AppData\Local\Temp\board_stubberpxc9nckz\stubs\micropython-v1_25_0-esp32-ESP32_GENERIC_S3
08:17:58|SUCCESS |mcu_stubber          - Stubs generated for esp32-ESP32_GENERIC_S3
08:17:58|SUCCESS |mcu_stubber          - Stubs copied to repos\micropython-stubs\stubs\micropython-v1_25_0-esp32-ESP32_GENERIC_S3
08:17:58|INFO    |mcu_stubber          - Merging stubs with docstubs : {'variant': '', 'build': '', 'arch': 'xtensawin', 'port': 'esp32', 'board': 'ESP32_GENERIC_S3', 'board_id': 'ESP32_GENERIC_S3', 'mpy': 'v6.3', 'ver': '1.25.0', 'family': 'micropython', 'cpu': 'ESP32S3', 'version': '1.25.0'}
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Python313\Scripts\stubber.exe\__main__.py", line 7, in <module>
    sys.exit(stubber_cli())
             ~~~~~~~~~~~^^
  File "C:\Python313\Lib\site-packages\rich_click\rich_command.py", line 378, in __call__
    return super().__call__(*args, **kwargs)
           ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Python313\Lib\site-packages\click\core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Python313\Lib\site-packages\rich_click\rich_command.py", line 166, in main
    rv = self.invoke(ctx)
  File "C:\Python313\Lib\site-packages\click\core.py", line 1728, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
              ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "C:\Python313\Lib\site-packages\click\core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python313\Lib\site-packages\click\core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "C:\Python313\Lib\site-packages\stubber\commands\get_mcu_cmd.py", line 94, in cli_create_mcu_stubs
    stub_connected_mcus(
    ~~~~~~~~~~~~~~~~~~~^
        variant=variant,
        ^^^^^^^^^^^^^^^^
    ...<4 lines>...
        bluetooth=bluetooth,
        ^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "C:\Python313\Lib\site-packages\stubber\bulk\mcu_stubber.py", line 391, in stub_connected_mcus
    merged = merge_all_docstubs(
        versions=board.firmware["version"],
    ...<3 lines>...
        mpy_path=CONFIG.mpy_path,
    )
TypeError: merge_all_docstubs() got an unexpected keyword argument 'mpy_path'

C:\Users\jon\AppData\Local\Temp\stubs>

Josverl, 2025-05-10:
Just as a sanity check 
,What is the output of `stubber --version`

jonhp, 2025-05-10:
C:\Users\jon\AppData\Local\Temp\stubs>stubber --version
micropython-stubber✏️ , version 1.24.4

C:\Users\jon\AppData\Local\Temp\stubs>

Josverl, 2025-05-10:
That is not the most recent version: 1.25.0.post3
https://pypi.org/project/micropython-stubber/

please reinstall / update to verify it solves the issue reported.

jonhp, 2025-05-10:
i guess i need more basic help:
i cleared pip cache
i uninstalled micropython-stubber
then I installed micropython-stubber
last few lines of cmd window:

Downloading micropython_stubber-1.24.4.post1-py3-none-any.whl (257 kB)
Installing collected packages: micropython-stubber
Successfully installed micropython-stubber-1.24.4.post1
PS C:\Users\jon\AppData\Local\Temp\stubs> stubber --version
micropython-stubber✏️ , version 1.24.4.post1
PS C:\Users\jon\AppData\Local\Temp\stubs>

Josverl, 2025-05-11:
ah that's my bad. 
stubber 1.25.x depends on mpflash 1.25.x , in which i found some significant issues and needed to yank from PyPI 
that is why pip falls back to an older version , which is not fixed 
uv explains the problem nicely : 
``` 
uv tool install micropython-stubber==1.25.0.post3
error: Because mpflash==1.25.0.post1 was yanked (reason: Defect in firmware-id  database upgrade) and only mpflash<=1.25.0.post1 is available, we can conclude that mpflash>=1.25.0.post1 cannot be used.
And because micropython-stubber==1.25.0.post3 depends on mpflash>=1.25.0.post1 and you require micropython-stubber==1.25.0.post3, we can conclude that your requirements are unsatisfiable.
```

as im not statisfield with the updates to mpflash 1.25.x, ill backport the fix to stubber 1.24.x  to get you unblocked

Josverl, 2025-05-11:
to summarize :
- micropython-stubber v1.24.4.post1 already has the fix.
- you reinstalled using pip , so it is installed , somewhere   
- when you run `stubber --version` you see a different version number , so that is a different installation that gets called.

Likely the cause is that stubber is not on your windows path. This normally gets handled by `acrtivate.ps1`, or `pipx install` or `uv tool install` 

To find which stubber.exe(s) are getting found type `where.exe stubber` in the command prompt. Then you'll get a lists of the installs that are found on your path that you may have overlooked when de-installing.

example output
```
> where.exe stubber 
D:\mypython\micropython-stubber\.venv\Scripts\stubber
D:\mypython\micropython-stubber\.venv\Scripts\stubber.cmd
c:\Users\josverl\.local\bin\stubber.exe
```
the first two are in a local virtual environment, the 3rd is installed by `uv tool install` or `pipx install`

my recommendation is to use 
- 'uv tool ensurepath'  Needed one time only to fixup your path for all tools  
- `uv tool install micropython-stubber`

[uv ](https://docs.astral.sh/uv/getting-started/installation/#__tabbed_1_2)is fast - and you do not need to worry about enabling an environment before using a tool.

jonhp, 2025-05-12:
Hi Jos,
I think I got there. How do I make the stubs available to IDEs like: PyCharm, VCS, Notepad++.
I would assume there are some instructions online somewhere that I didn't find...
THANKS, Jon
==============================================================

`Microsoft Windows [Version 10.0.26100.3775]
(c) Microsoft Corporation. All rights reserved.

C:\Users\jon>d:

D:\>cd BitBucket\HT16K33-Python

D:\BitBucket\HT16K33-Python>ls
LICENSE.md  README.md  examples  ht16k33  ht16k33.code-workspace  package.json  pyproject.toml  tools

D:\BitBucket\HT16K33-Python>uv tool install micropython-stubber==1.25.post3
  x No solution found when resolving dependencies:
  `-> Because mpflash==1.25.0.post1 was yanked (reason: Defect in firmware-id  database upgrade) and only
      mpflash<=1.25.0.post1 is available, we can conclude that mpflash>=1.25.0.post1 cannot be used.
      And because micropython-stubber==1.25.0.post3 depends on mpflash>=1.25.0.post1 and you require
      micropython-stubber==1.25.post3, we can conclude that your requirements are unsatisfiable.

D:\BitBucket\HT16K33-Python>stubber --version
micropython-stubber✏️ , version 1.24.4.post1

D:\BitBucket\HT16K33-Python>where.exe stubber
C:\Python313\Scripts\stubber.exe

D:\BitBucket\HT16K33-Python>uv tool ensurepath
Executable directory C:\Users\jon\.local\bin is already in PATH

D:\BitBucket\HT16K33-Python>uv tool install micropython-stubber
Resolved 128 packages in 4.41s
      Built esptool==4.8.1
      Built blkinfo==0.2.0
Prepared 127 packages in 8.21s
Installed 128 packages in 7.87s
 + annotated-types==0.7.0
 + ansicon==1.89.0
 + anyio==4.9.0
 + argcomplete==3.6.2
 + argparse-addons==0.12.0
 + attrs==25.3.0
 + autoflake==2.3.1
 + beautifulsoup4==4.13.4
 + bincopy==20.1.0
 + bitarray==3.4.0
 + bitstring==4.3.1
 + black==24.10.0
 + blessed==1.21.0
 + blkinfo==0.2.0
 + build==1.2.2.post1
 + cache-to-disk==2.0.0
 + cachecontrol==0.14.3
 + cachetools==5.5.2
 + certifi==2025.4.26
 + cffi==1.17.1
 + charset-normalizer==3.4.2
 + cleo==2.1.0
 + click==8.2.0
 + colorama==0.4.6
 + crashtest==0.4.1
 + cryptography==44.0.3
 + deprecated==1.2.18
 + distlib==0.3.9
 + docutils==0.21.2
 + dulwich==0.22.8
 + ecdsa==0.19.1
 + editor==1.6.6
[... 664 lines cut ...]
micropython-v1_25_0-samd-SEEED_WIO_TERMINAL-merged
micropython-v1_25_0-stm32-PYBV11
micropython-v1_25_0-stm32-PYBV11-merged
micropython-v1_26_0_preview-docstubs
micropython-v1_9_3-esp8266
micropython-v1_9_3-frozen
micropython-v1_9_4-esp8266
micropython-v1_9_4-frozen
pycopy-v0_0_0-frozen

D:\Users-Data\Jon\Documents\espressif\esp32-s3\repos\micropython-stubs\stubs>cd micropython-v1_25_0-esp8266-ESP8266_GENERIC-merged

D:\Users-Data\Jon\Documents\espressif\esp32-s3\repos\micropython-stubs\stubs\micropython-v1_25_0-esp8266-ESP8266_GENERIC-merged>ls
_espnow.pyi    doc_stubs.json       gc.pyi           mip           select.pyi     ubluetooth.pyi  usocket.pyi
_onewire.pyi   ds18x20.pyi          hashlib.pyi      neopixel.pyi  socket.pyi     uerrno.pyi      ussl.pyi
apa102.pyi     errno.pyi            heapq.pyi        network.pyi   ssd1306.pyi    uio.pyi         ustruct.pyi
binascii.pyi   esp.pyi              inisetup.pyi     ntptime.pyi   time.pyi       ujson.pyi       usys.pyi
btree.pyi      espnow.pyi           lwip.pyi         onewire.pyi   tls.pyi        umachine.pyi    utime.pyi
cryptolib.pyi  firmware_stubs.json  machine.pyi      platform.pyi  uarray.pyi     uos.pyi         uzlib.pyi
deflate.pyi    flashbdev.pyi        math.pyi         random.pyi    uasyncio.pyi   uplatform.pyi
dht.pyi        framebuf.pyi         micropython.pyi  requests      ubinascii.pyi  uselect.pyi

D:\Users-Data\Jon\Documents\espressif\esp32-s3\repos\micropython-stubs\stubs\micropython-v1_25_0-esp8266-ESP8266_GENERIC-merged>notepad machine.pyi

D:\Users-Data\Jon\Documents\espressif\esp32-s3\repos\micropython-stubs\stubs\micropython-v1_25_0-esp8266-ESP8266_GENERIC-merged>`

Josverl, 2025-05-13:
wrt formatting -  Please use triple  backtics aka codeblocks ```  for things that are more than a line, and /details for large blocks to keep things readable 

## how to use / install ?

```
                                                       Results
┏━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━┓
┃ Result   ┃ Name/Path                                                                        ┃ Version      ┃ Error ┃
┡━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━┩
│ Build OK │ micropython-esp32-esp32_generic_s3-stubs                                         │ 1.25.0.post1 │       │
│          │ repos/micropython-stubs/publish/micropython-v1_25_0-esp32-esp32_generic_s3-stubs │              │       │
└──────────┴──────────────────────────────────────────────────────────────────────────────────┴──────────────┴───────┘
10:58:52|SUCCESS |mcu_stubber          - Done

```

Shows that your custom stubs have been stored in : `repos/micropython-stubs/publish/micropython-v1_25_0-esp32-esp32_generic_s3-stubs`

you can 
 - copy this folder somewhere for safekeeping,
 - install the stubs from that folder by specifing that as the source to install from, rather than pull the package from PyPI
    - `uv pip install --target typings repos/micropython-stubs/publish/micropython-v1_25_0-esp32-esp32_generic_s3-stubs` 
    - `pip install --target typings --no-user repos/micropython-stubs/publish/micropython-v1_25_0-esp32-esp32_generic_s3-stubs`

A few questions if I may ?
- is this a custom board or build ? 
- if so, what is different ?
- would it be useful to name the stub package different , such as `micropython-esp32-myspecials3-stubs` ?
- would it be useful to have stubber copy the result in a different location ? so instead of `repos/micropython-stubs/publish/..` store   it in the current folder or a folder you specify ?