Josverl/micropython-stubber #667: github.GithubException.BadCredentialsException: 401

kuyeon, 2025-05-20:
**Describe the bug**
github.GithubException.BadCredentialsException: 401 Error occurs when I execute the command below.

`stubber clone --add-stubs`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
`2025-05-20 15:11:55.866 | INFO     | stubber.utils.config:readconfig:91 - Could not find config file: pyproject.toml
2025-05-20 15:11:56.324 | WARNING  | stubber.utils.config:post_read_hook:76 - Could not read micropython versions from git: 401 {"message": "Bad credentials", "documentation_url": "https://docs.github.com/rest", "status": "401"}
Traceback (most recent call last):
  File "/home/k/Develop/railrobot_v3/.venv/bin/stubber", line 5, in <module>
    from stubber.stubber import stubber_cli
  File "/home/k/Develop/railrobot_v3/.venv/lib/python3.11/site-packages/stubber/stubber.py", line 20, in <module>
    from stubber.commands.switch_cmd import cli_switch
  File "/home/k/Develop/railrobot_v3/.venv/lib/python3.11/site-packages/stubber/commands/switch_cmd.py", line 21, in <module>
    VERSION_LIST = git.get_tags("micropython/micropython", minver="v1.9.3") + ["latest"]
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/k/Develop/railrobot_v3/.venv/lib/python3.11/site-packages/cachetools/_decorators.py", line 17, in wrapper
    v = func(*args, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^
  File "/home/k/Develop/railrobot_v3/.venv/lib/python3.11/site-packages/stubber/basicgit.py", line 127, in get_tags
    gh_repo = GH_CLIENT.get_repo(repo)
              ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/k/Develop/railrobot_v3/.venv/lib/python3.11/site-packages/github/MainClass.py", line 363, in get_repo
    headers, data = self.__requester.requestJsonAndCheck("GET", url)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/k/Develop/railrobot_v3/.venv/lib/python3.11/site-packages/github/Requester.py", line 442, in requestJsonAndCheck
    return self.__check(
           ^^^^^^^^^^^^^
  File "/home/k/Develop/railrobot_v3/.venv/lib/python3.11/site-packages/github/Requester.py", line 487, in __check
    raise self.__createException(status, responseHeaders, data)
github.GithubException.BadCredentialsException: 401 {"message": "Bad credentials", "documentation_url": "https://docs.github.com/rest", "status": "401"}`

**Stubber information **
 - Version : latest
 - run directly or though Micropy-cli : directly
  
**Firmware Information **
 - Family: micropython
 - Version 1.19.*
 - specific additional modules ( if any) : esp32 
 
** Hardware information **
 - Make & version
 - CPU 
 - Link to HW spec ( if possible) 
 
 
**VS Code Extension Information (please complete the following information):**
 - OS: Ubuntu22.04

**Additional context**
Add any other context about the problem here.

Is this git token expired?

Josverl, 2025-05-20:
Thanks for the report.
I agree that the error message should be clearer...

I indeed include a RO GitHub token to avoid rate limits, so I'll check that for expiration.

Can you confirm the version is micropython-stubber==1.25.0.post3 ?

Josverl, 2025-05-21:
I tried to install micropy-cli on windows and ubuntu,  and in both case the package is not installed correctly.
That error should be reported in the micropy-cli repo.  

I do notice that micropy-cli has [a pinned dependency on an old version of stubber](https://github.com/BradenM/micropy-cli/blob/f361a0465c155c11be66da3515358ca25e79dc3e/pyproject.toml#L68C1-L68C33), for which the token has likely expired.

my recommendation is to upgrade to a more recent version of stubber.
``` 
pip install -U  micropython-stubber
pip show micropython-stubber
```
 

P.S. Since a recent dependency update stubber shows some warnings on shorthand parameter names, but they do not prevent stubber from working, so you can ignore them for now. #669

closed 2025-06-11 by Josverl